tinypico/tinypico-hardware #3: How are the values of L1, C4 and C5 decided?

dataplayer12, 2019-10-06:
Hi, I am wondering how you decided to use the values of L1 (7.5 nH), C4 (1.5 pF) and C5 (0.75 pF) for use with the antenna? The datasheet for the ESP32-Pico-D4, these values are different. Which value of these three would the performance be most sensitive to?

UnexpectedMaker, 2019-10-06:
The values on a matching network are specific to the exact board and layout its on. The values on the PicoKit are ONLY for the PicoKit - the values for the TinyPICO are ONLY for the TinyPICO.

Everything about the components on the board, their positions are orientations, the size of the board, number of layers, power and gnd copper pours, type of antenna, orientation of antenna, what other metal is near it.. it all changes the RF characteristics of the board.

The values on the TinyPICO are based on having it professionally matched by an RF expert.

Sounds like you need to do lots of reading up about RF and matching networks :)

dataplayer12, 2019-10-06:
Thanks a lot. I would really appreciate any pointers to reading material.

nevercast, 2019-10-07:
@dataplayer12 We do a bit of RF tuning at work, and what I have found is that, provided the antenna is 50ohms matched, you can usually put a 0ohm resistor across the series element (the series element on this particular board is C5, the element that is in series with the antenna line).

@UnexpectedMaker is correct that everything will change the performance of your antenna, and usually its best to tune it for its normal condition. For example, the TinyPICO probably spends most its time in a breadboard, so perhaps that would be the best situation to tune the antenna.

For your own board, it might usually be inside a chassis, and if its a handheld device, maybe it spends a lot of time in someones hands. So perhaps its best to tune it for that.

The fact is, your tuning will always be influenced by the environment and its not going to be perfect when its being used if that situation differs from when it was tuned in a lab. What advice I can give is that, if you're unable to follow recommended layout guidlines set out by the antenna you're using (such as shape and size of copper fills, and the proximity to other components), you might want to consider getting it tuned. If however, you have a very short trace from antenna port on the ESP32 to antenna on your board, you're probably fine putting a 0ohm series element in there, since the antenna should be already pretty close to tuned.

If you find the need to properly tune your antenna then I suggest buying a VNA, since unless you have a friend that will do the tuning for free, hiring an RF engineer with the equipment can be very experience. We use a miniVNA at my place of work and it does the job.

One extra point, when tuning, we usually put the tuned frequency at the left hand side -3dB point, so that the tuned frequency is slightly higher than ideal. That's because we suspect the device to be loaded by its surroundings and that will pull the frequency down to its ideal point.

I'd be happy to discuss this sometime if you want more information, find me on Twitter https://twitter.com/NvrCast

dataplayer12, 2019-10-07:
Thanks. Your explanation is very helpful. That explains why in some boards, people have left out the values of parallel elements. There may be no need to use them for most applications. I am curious as to why a 0 ohm series element is necessary. After all it is 0 ohm and as far as I understand, electrically it does nothing.

UnexpectedMaker, 2019-10-07:
The 0 ohm bridges he pads... otherwise there is a short between the ESP32 and the Antenna

dataplayer12, 2019-10-07:
Yes, I understand that, but could you not eliminate this part entirely and just have a wire in place of the pads.

nevercast, 2019-10-07:
> and as far as I understand, electrically it does nothing.
Correct. As @UnexpectedMaker pointed out, without briding the gap, the antenna and ESP32 are not connected. So you place a 0Ohm component in what would be C5 for the TinyPICO and that at least connects the Antenna to the ESP32 antenna port.

UnexpectedMaker, 2019-10-07:
Sure, you could just put a trace instead of the pads, but then how do you tune it with parts? The pads are there so you (if you get the expertise) or someone else can tune it... if you don't care about good quality RF, or even more, if you don't care that the implementation of your antenna could be reflecting, causing unwanted heat and power drain, then sure.. don't put any matching network, and just stick an antenna on there :)

UnexpectedMaker, 2019-10-07:
Unfortunately, RF is complicated, and it's not a "one size fits all" - every board design that implements an antenna should be tested to *at least* ensure the board is performant and doesn't have adverse effects from heat or power.

Ultimately, if you are making a hobby project for yourself, and putting an antenna on your board gives you the wifi quality you need, you can avoid getting it matched.., but if you are making something for others to use, then a properly matched network is the right thing to do.

dataplayer12, 2019-10-07:
Your advice makes a lot of sense. Since this is my first board, I will not tune it and see how it works. Thanks, you have been a great help!

closed 2019-10-07 by dataplayer12

niemeyer, 2021-02-06:
Here we are in 2021, on a Saturday, and this is still an interesting thread. Kudos to all commenters.

Montvydas, 2021-04-25:
I know the issue was closed, but since the TinyPico changed the RF design slightly, I was wondering, why was the matching network removed from V2 design completely as can be seen [here](https://github.com/tinypico/tinypico-hardware/blob/master/TinyPICO%20V2%20Schematics/TinyPICO_V2_C2_Schematic.pdf)? This surprised me since the previous design later added an identical 3D antenna, thus I was expecting to see a similar matching system..

matt-42, 2022-01-07:
> I know the issue was closed, but since the TinyPico changed the RF design slightly, I was wondering, why was the matching network removed from V2 design completely as can be seen [here](https://github.com/tinypico/tinypico-hardware/blob/master/TinyPICO%20V2%20Schematics/TinyPICO_V2_C2_Schematic.pdf)? This surprised me since the previous design later added an identical 3D antenna, thus I was expecting to see a similar matching system..
> 
> Edit: So it seems the matching network should be removed if the transmission line length is smaller than 5mm (1/10 of wavelength), I guess that might be the case then on V2...

That would be great where did you find this info? I found another ESP32 board without matching circuit: https://github.com/TobleMiner/ESP32-Devkit-Type-C/issues/3

Montvydas, 2022-01-13:
> That would be great where did you find this info? I found another ESP32 board without matching circuit: [TobleMiner/ESP32-Devkit-Type-C#3](https://github.com/TobleMiner/ESP32-Devkit-Type-C/issues/3)

Well, the bits of information are scattered there and there, but if you watch [this talk](https://www.youtube.com/watch?v=ZYUYOXmo9UU), or at least the bit starting at 38:30, it talks about when to care about the line lengths. In the video they used 1/12 of the wavelength, which is more precise, but 1/10 is easier to calculte, that's why it's usually used... So a rule of thumb! Basically a typical PCB dielectric (FR-4) has a relative permittivity of ~4.5. If  it's an inner layer trace, then the refractive index is 2.12 (square root of permittivity). On the outer layer, it is a combination of both air and the PCB dielectric, thus the relative permittivity of ~3.5, then the refractive index is 1.87. Thus, depending on the scenario the velocity drops by a factor of 2.12 (inner) / 1.87 (outer), which changes the wavelength of the 2.45GHz signal to 57.7mm / 65.5mm (wavelength = speed / freq). 1/12th of that is 4.8mm / 5.45mm, meaning if the transmission line is longer than that, then a matched network needs to be added. This is theoretical though, since any components on the board and the added device case (even if made of plastic) will affect the antenna and thus it will require rematching.. However with a case on is added the BLE will still work, it will simply add losses and thus battery will be drained somewhat faster if you want to keep the same signal strength. 

Anyway, an easy thing to do is to add a pi-network with a 10pF capacitor shorting the MCU and the antenna. At 2.45GHz 10pF is somewhat close to 0 Ohm impedance and thus it will not change anything. However if you later decide that you want you device matched you can send it to a specialised service and they will decide what the pi-network components should be. Note that a 0 Ohm resistor will not work as 0 Ohm impedance at 2.45 GHz, since it has inductance, thus better use the 10pF capacitor (it is worth checking manufacturer graphs for their capacitors as they got impedance vs frequency graphs for various capacitor sizes and packages.. thus helping to decide for the best value, just 10pF is usually somewhat OK). Hope that helped!

Montvydas, 2022-01-14:
Regarding TobleMiner/ESP32-Devkit-Type-C#3, if you use a PCB antenna, then you can avoid using a matching network by firstly making the antenna tail longer than needing. Then during the matching process you can cut the tail bit-by-bit until you establish a matched network. This type of process changes the antenna built-in impedance (combination of capacitance and inductance) by physically altering the antenna rather than introducing extra components. That's one of the benefits for using a PCB antenna in this case, but you would still need a VNA (or can get a nanoVNA for under £50 on Aliexpress). I think I learnt this from Robert Feranec video, where they discussed antennas, but cannot be 100% sure now, it was some time ago now...

matt-42, 2022-02-08:
Thanks @Montvydas for all the explanations :)