tinypico/tinypico-hardware #5: how much current will the gpio35 draw?

dukeduck1984, 2019-11-26:
when the battery voltage ADC is not in use, how much current, if any, will the gpio35 draw?

Thanks,
Kaiyuan

nevercast, 2019-11-26:
GPIO35 is an input only pin and has no hardware drivers. Therefore the pins should follow nominal leakage characteristics when not being sampled. Therefore I would expect 50nA according to the input current specified here: https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf#page=41 and here https://www.espressif.com/sites/default/files/documentation/esp32-pico-d4_datasheet_en.pdf#page=15 

[image]

nevercast, 2019-11-26:
Looking at the schematic https://github.com/tinypico/tinypico-hardware/blob/master/TinyPICO%20Release%20P3/TinyPICO_Schematic.pdf

The divider is always enabled when on battery and sends the battery to ground through about 600kOhm which is about 6 microamps at 3.7V. You can probably increase the resistor values and keep their ratio if 6 microamps is too much for you.

UnexpectedMaker, 2019-11-26:
The divider is always on when the board is powered, even if no battery is present, as the PMIC can't tell the different between no battery and a full battery, so still tries to charge, so VBAT always has voltage, so the divider is always sucking current.

But as Josh pointed out it's low uA... I'd not increase the resistors unless you don't want to be able to read the voltage accurately, as higher value resistors will just increase their susceptibility to noise, decreasing the ADC's accuracy.

So if you don't need to read the voltage, removing one of the divider resistors means no current use at all, but if you do want to read the voltage, ultimately, it's not costing much.

The TinyPICO deep sleeps at 18uA or less.

closed 2020-03-17 by UnexpectedMaker