tinypico/tinypico-hardware #7: Where can I download a 3D Model?

Pro, 2019-11-28:
We would need the 3D Model of the TinyPico for building a Case and visualizing the board on it.

Could you add the model to the git repo (in the ideal case as STEP file)? Or is it available somewhere else?

UnexpectedMaker, 2019-11-28:
Sorry, there is no 3D model for the TinyPICO.

Seon Rozenblum | Unexpected Maker
http://unexpectedmaker.com <http://unexpectedmaker.com/>

Pro, 2019-11-28:
Bad news, but thanks anyways!

closed 2019-11-28 by Pro

taf2, 2021-12-15:
I found this one which is pretty good https://grabcad.com/library/tinypico-esp32-1  it is an older model but I've been using it with some success

UnexpectedMaker, 2021-12-15:
Official models are all available here.... :)
https://github.com/tinypico/tinypico-hardware/tree/master/3D%20Files