tinypico/tinypico-hardware #8: External 5V Power supply and USB 5V at the same time?

Pro, 2019-11-28:
I have a setup where the 5V Pin is connected to an external power supply.

My question:
Is it safe to plug in the USB Cable with its own 5V Power Supply while there's still 5V coming through the 5V Pin?

Use Case is that I need to re-flash it while the external power supply is on, or I want to open the monitoring through USB.

Thanks!

Pro, 2019-11-28:
I already checked the board itself and saw that the 5V goes to a Chip with labeling "65b902". I couldn't find out which chip it is.

UnexpectedMaker, 2019-11-28:
No, there is no reverse USB protection, so you may backfeed power and damage your host USB port. Only way is to ensure the power going into the 5V pin is lower than the 5V from the USB, but even then, its not recommended.

Also, this repo is not for TinyPICO support, sorry...  Please join my discord server for support Q's.
https://discord.gg/xAHpApP

Thanks.

Pro, 2019-11-28:
Ok, sorry for spamming here. Will use the Discord Server! Thanks anyway for answering!

closed 2019-11-28 by Pro