tinypico/tinypico-hardware #9: Low-Battery / Over discharge

CrispinP, 2020-02-19:
What stops over discharge of the battery?

I can't see anything in the schematic that would handle this.

[image]
The ESP can operate down to 2.2V. With the LDO drop of 190mV that would pull the battery down to 2.3/2.4V before it would get flaky. A bit low for a li-ion battery.

Have I missed something?

nevercast, 2020-02-19:
For what it's worth, the ESP32-PICO can only work down to 2.7V due to the internal flash being a 3v3 device. You will get flash errors lower than 2.7V

UnexpectedMaker, 2020-02-19:
Yup, you need to use protected LiPos. The ones with the yellow on the top.
This really isn't a TinyPICO github issue, and is more a question for Discord :)

CrispinP, 2020-02-19:
ah ok. Thanks.
So I'm not reading it wrong.

(Agreed on the last point 😄 )

closed 2020-02-19 by CrispinP