tinypico/tinypico-hardware #10: Req: Missing part libraries

henrygab, 2020-03-17:
Just found and picked up my first TinyPICO ... very impressive form factor.

I went ahead and loaded the board & schematics in Fusion 360 (which now has the functionality of Eagle built-in).  The problem is, many of the parts used reference libraries that are not public.

Could you please do one of the following two things?

1. make the following libraries, which contain parts used by the TinyPICO, available in the git repo?
  * cmonr_standardparts
  * CP2104
  * microbuilder
  * NCP170AMX300TCG
  * Philbywhizz
  * supply1
  * UnexpectedMaker

2. Alternatively, can you put all the parts used for this board into a single library, and make it a "managed" library in Eagle? This would allow it to be downloaded automatically by Eagle / Fusion360 users, including getting updates if anything changes.

Having the libraries for the parts is the first step to enabling full 3D models in Fusion360.  Having the 3D model makes it **_really_** easy for makers to create custom enclosures, using any output supported by Fusion360 (e.g., 3D-printing, laser-cut acrylic, CNC, etc.).  The 3D model it makes is already really good, but it can't show the usb connector, and the boxes are wrong-sized for other parts.  Still, it's already really close!

The only remaining step then would be to ensure the parts have the 3D models ... which is itself a zero-cost service (for end users).  Mouser, for example, has some relationship with [LibraryLoader](https://www.samacsys.com/library-loader-help/), and they turnaround simple new requests (e.g., common chip packages) in ~48 hours.

Thanks!

UnexpectedMaker, 2020-03-17:
Hi Henry,

I didn't supply a TinyPICO library because it's really not needed due to the way Eagle works.

All of the data you need for each part on the board is in the board file, so you can easily and quickly make your own library of the parts of you want to go down the road of attaching 3D models to the components.

Create a library, open it in the editor, and then right click on the parts you want to add, and copy the device to the library. From there, you can assign 3D models if you choose.

As the project is open source, what you are asking for seems like a perfect candidate for what you (or others could make) and then contribute back to the project :)

Cheers,
Seon

henrygab, 2020-03-18:
Hi @UnexpectedMaker ,

Thanks for the quick response.  I am willing to do this, and submit a PR.  The steps for doing this are not obvious.

<details><summary>Parts list context menu fails...</summary><P>

However, I'm not entirely clear how to copy a device that has no backing library from and existing brd/sch pair.

My steps to try:
1. Create new user library
2. Open the TinyPICO.sch schematic in Eagle
3. Select `Design Manager` super-tab
4. Select `Browser` to see the parts list
5. Click on header `Device` to sort by device name
6. Right-click on part `IC2G$1`, the CP2104 USB-to-Serial chip
  a. Clicking `Package` results in error: `Part 'IC2G$1' not found!`
  b. Clicking either `Open Symbol` or `Open Device` results in error: `Library CP2104.lbr was not found in current library path(s). Please adjust the library paths or download or export the drawing libraries first!`

</details>
<details><summary>Export?</summary><P>

I am now trying File->Export, to dump all the parts used into a new library.  This seems to generate a large script, and runs the script to add parts to the new library ... with all the performance that suggests.  I'm waiting for this to finish. 
</details>

Is this export option the start of the path you were suggesting, or did you have another alternative in mind?  I couldn't find a way to just right-click on a part to be able to add it to a new library....

UnexpectedMaker, 2020-03-18:
1. Create and open a new library.
2. Import & open TinyPICO schematic
3. Right click on a symbol in the actual schematic
3. Select "Copy device to library" - will be greyed out if the library is not open. 
4. Repeat for all components, then save the library.
5. Win!

This is the Eagle workflow that also works in Fusion360

[image]

henrygab, 2020-03-18:
Awesome!  Thanks for the pointers.  I'll see what I can put together!

closed 2020-03-18 by henrygab

henrygab, 2020-03-18:
Update:  Any efficient use of Eagle appears beyond my ken.  I've either given up, or put this on a backburner for now.