tinypico/tinypico-hardware #16: tinypico board layout

bilaljo, 2021-06-24:
I would like change out the prog resistor R3 against another (to change the battery current properly for my LiPo). Since I have only the schematic (and real board) it is hard to figure out which one is the correct one. I could only guess, which is quite unsafe.
Could you push an layout file of tiny pico (nano) as well?
Thanks.

UnexpectedMaker, 2021-06-24:
Sorry, which board are you asking about the VBAT Prog resistor for? The TinyPICO or the TinyPICO Nano?

bilaljo, 2021-06-24:
Sorry vergot to mention. The TinyPICO Nano.

UnexpectedMaker, 2021-06-24:
I've highlighted the resistor with a blue outline.

[image]

bilaljo, 2021-06-24:
Great, that's what I was searching for! Thank you very much.

UnexpectedMaker, 2021-06-24:
Nps!

closed 2021-06-24 by UnexpectedMaker