tinypico/tinypico-hardware #18: Other I2C pin than IO21 and IO22

bilaljo, 2021-07-08:
It is possible to use other pin than IO21 and IO22 for I2C? They are marked as I2C pins on the tinpy pico pinout which makes me quite unsure.

UnexpectedMaker, 2021-07-08:
You can initialise I2C with any pins, but 21,22 will give you hardware I2C, where other pins will be software I2C.

bilaljo, 2021-07-08:
Thank you for your answer. Is that a tiny pico specific feature? Since I found this post on the esp32 forum:

"Yes, the digital portion has two hardware I2C modules. You don't see the pins for it in the pinout because they're conencted to the GPIO mux and you can connect them to any of the (output-capable) GPIOs."

https://esp32.com/viewtopic.php?t=215

UnexpectedMaker, 2021-07-08:
TinyPICO is an ESP32 and can do whatever an ESP32 can do. Some pins will give you hardware, some software I2C depending on which you use. 

The ESP32-PICO-D4 datasheet is the best place to look.

closed 2021-07-16 by UnexpectedMaker