tinypico/tinypico-hardware #20: Unable to flash tiny pico nano

bilaljo, 2021-07-15:
Is there anything to consider when flashing a self-designed board with a tiny pico nano?
I am using a micro usb B as port and UART is working fine (since I am able to see the tiny pico nano as listed device) but I got always the error
"esp32 invalid head of packet"
I know thats for esp32 a well-known issue but all suggested solutions do not working (especially because they are only for pre-builded boards).

UnexpectedMaker, 2021-07-15:
Do you have anything connected to IO0, IO2, IO5 or IO12?
Can you share your schematic with me? You can email it to me if you don't want to post it here publicly.

All Nanos go through the testing twice. Once before being cleaned and once again after being cleaned, so all boards have successfully been flashed 2 times with MP and have had all of the tests run on them twice.

bilaljo, 2021-07-16:
Yes, I have IO12 connected to a device.

UnexpectedMaker, 2021-07-16:
IO12 is a strapping pin.  It needs to be untouched at boot, or at best, never pulled high during boot.
Check the PICO-D4 Datasheet for all scraping pins and what you can and can't do with them during boot up.

bilaljo, 2021-07-16:
Thank you very much. In fact it was a socket for a battery so only connecting + and - (without battery) while flashing solved it.
Oh I see I meant actually IO2 but I think same problem with IO2, right?

UnexpectedMaker, 2021-07-16:
Why are you connecting a battery to IO2? Never connect a power source to an IO pin.

bilaljo, 2021-07-16:
For watching the battery level since it has an ADC.
Oh is that a serious problem?

UnexpectedMaker, 2021-07-16:
So long as it’s going through a voltage divider it’s ok.

You should have nothing connected to any strapping pins at all that influence the state of them at boot time.

None of this is TinyPICO Nano related, so I’m going to close this as it’s not an issue.

Feel free to join my discord to ask questions there.

closed 2021-07-16 by UnexpectedMaker