tinypico/tinypico-hardware #29: PCB Dimensions for footprint and Shield Ground

JL1946, 2024-07-01:
Hello...
I am making a Kicad footprint for the TinyPICO P3 Board... I am using the tinypico/tinypico-hardware/TinyPICO Release P3/Eagle/TinyPICO.brd to make the measurements using the built-in Kicad measuring tool... With the board placed such that the long-side is in the  y-direction and the short side in the x direction, the distance between the each pad in the y-direction is the standard 2.54 mm or .1 in... However, in the X direction I am measuring 15.2654 mm or 0.601 in between the pads (center-to-center)... Is this correct??? As for the size of the board (less critical), I am measuring the y-direction as 31.852 mm or 1.254 in. and 17.780 mm or  0.7 in.  Are these measurements in the ball-park???

I have included some jpg photos to show the orientation of the footprint... 

One last thing... From the schematic, I saw that you tied the shield of the USB-C connector to the power/signal ground... I am incorporating your board onto another board which implements a separate trace for Earth-Ground which the usb-c cable shield routes to an Earth connection at the power supply... The design is using TVS diodes to transfer any ESD picked up by the sensors of the system to that Earth-Ground path... However, with Shield tied to the signal/power ground, it seems to me that any ESD would be routed to that ground as well... What is your opinion on this issue? Thanks for your timely response on the issues that I have outlined in this writeup...
[image]
[image]

UnexpectedMaker, 2024-07-01:
I guess all that really matters here is to have 32x18mm clearance for the board (maybe longer to ensure antenna clearance from other components on your carrier board) and the headers are 17.78m apart (7x 2.54mm header pin spacing).

With regards to the USB shielding being connected to GND, I'm not really sure how you have all of that connected as you mention another USB-Connector connected to your power supply?

Even without the TinyPICO USB Shield connected to the TinyPICO GND, you still have GND connected via the TP USB-C connector through the wires, so that is already going to connect to your "Earth GND" if you have them common - and I'm. not sure how it would work if they were not common - technically both could be the same value, but technically they could be quite different, and then you'd mess with the voltage rails on the TinyPICO.

I think what you need to do, is if you are connecting USB to the TinyPICO (I assume for flashing etc, but other times the power will come from your other supply?) then you should use a modified USB-C connector that has GND and maybe 5V cut so it only provides D+/D-.

The USB-C shield -> GND part really makes no difference here as You are connecting USB-C->GND anyway.

Without seeing your intended schematic or having a thorough understanding of what you are doing I can't really advise any best practice here, sorry.

JL1946, 2024-07-05:
Hi...
Thank you for your response... Please clear this up for me... I measured the distance between the headers to be 6 x 2.54 mm = 15.24 mm, not 7 x 2.54 = 17.78 mm... I measured the board width to be 17.78 mm which would make sense if you list the pcb as 18mm x 32mm... 

Sorry to hound you, but I live in Canada and am ordering the TinyPICO V3 from the USA... My pcb's are being made in China... I don't want to make a mistake... thanks for understanding...

UnexpectedMaker, 2024-07-05:
Sorry, I don't understand the question, you don't measure between the headers, you measure from header to header, The headers are 17.78mm apart.

How much space you have between them is not a measurement I can give you, as it depend on how you mount the board to your carrier., What type of headers you use. Surface mount headers on your carrier will take more space then through hole.

JL1946, 2024-07-06:
I have included 3 photos to try to clarify my issue... The first is a picture of your board I want to use... the second is a jpeg of the motherboard I designed that shows the footprint placement of the TinyPICO P3 board that I want to buy from you... The third is a jpeg of the Top Copper Layer or TinyPICO.cmp file from your Github/.../TinyPICO Release P3/Gerbers_274X directory... This jpeg shows 3 pads marked "A", "B", and "C"... According to the Gerber Viewer that I am using, The x-y positions of the pads in mm (relative to 0,0 axis) are as follows:
"A" --> x:4.8885  y: 21.3444
"B" --> x:7.4285   y: 21.3444
"C" --> x:4.8885  y: 6.0790 

When I do the math, I get the distance between Pads "A" and "B" to be 2.54 mm... The distance between Pads "A" and "C" to be 15.2654 mm which is very close to 6 times 2.54, NOT 7 times 2.54... The Gerber viewer I'm using can't be wrong or, otherwise the calculated distance between Pads "A" and "B" would be wrong...

I want to have my board manufactured but don't want to make a mistake and waste time and money... I hope that this clarifies the issue I'm having...

Also, If I purchase from your website:
( https://unexpectedmaker.com/shop.html#!/TinyPICO/p/577111313/category=154494282 ) Do you ship to Canada?

[image]
[image]
[image]

UnexpectedMaker, 2024-07-06:
Sorry, I miss understood what you were asking originally.

Here are the dimensions in an easy to understand way:
[image]

Yes, I ship all over the world from my website :)